Feilner/homebridge-plc #13: Connection to PLC failed

Nrigoni, 2020-10-31:
Hello Feilner, 

I need your help to understand what I'm doing wrong in connecting the PLC to homebridge PLC ip is 192.168.10.1 and homebridge is connected to wifi house that has another ip adress I suppose that this is the problem right? but can you help me fixing this?

Thanks for your support

Feilner, 2020-10-31:
Hello Nrigoni,

I have not understood your network setup completely maybe you can post network setup.:
IP, subnetmask and Gateway for each device.

Otherwise I can only help with general statements. 

The connection from iPhone to Homebridge or if you have a Apple TV or HomePod as ControlCenter the connection from the ControlCenter to Homebridge works only within the same sub net. The protocol is not routable (but there seems also to be some workarounds)

The connection from Homebridge to the PLC (s7 Protocol) is routable. It works also in different subnets. But in order that it works correctly that requires that. the IP, subnetmask and Gateway is correctly set at the host that runs Homebridge as well as at the PLC. Also your gateway/router needs to have set the route correctly. A good test is if you try to **ping** your PLC from the homebridge terminal.
 
Which PLC are you using have you that the rack and slot number correctly?

Best regards,

Feilner

Nrigoni, 2020-10-31:
thanks fo the quick reply I'm doing a lot of test .. hope fide the solution so I try to put here all the network configuration 

PLC in tia porta I assigning IP 192.168.10.1
PC windows connected to PLC in the IPv4 protocol 
IP:192.168.10.80 subnet 
mask:255.255.255.0 
default gateway: left empty
Preferred DNS server: left empty
Alternative DNS server: left empty
In this same PC I run homebdrige UI over chrome it connect to home network WiFi with this IP 192.168.1.43

Hope this is clear... wait your suggestion on how to proceed

Feilner, 2020-11-01:
Have you tried to ping the PLC from Homebridge UI Terminal. (Crock on the three dots and select terminal)

Which PLC are you using?
Have you checked that the rack and slot number is correctly in the configuration?

Nrigoni, 2020-11-01:
I don't have the terminal option in the Homebridge UI , I'm using an S7-1200

this is the config.json

{
    "bridge": {
        "name": "Homebridge D433",
        "username": "0E:88:80:0A:D4:33",
        "port": 51906,
        "pin": "913-35-705"
    },
    "accessories": [],
    "platforms": [
        {
            "platform": "PLC",
            "ip": "192.168.10.1",
            "rack": 0,
            "slot": 1,
            "enablePolling": true,
            "enablePush": true,
            "enableControl": true,
            "port": 8080,
            "accessories": [
                {
                    "accessory": "PLC_LightBulb",
                    "name": "LuceSala",
                    "manufacturer": "normal light bulb",
                    "db": 1,
                    "get_On": 0,
                    "set_On": 0.1,
                    "set_Off": 0.2
                }
            ]
        },
        {
            "name": "Config",
            "port": 8581,
            "platform": "config"
        }
    ]
}

and this is the setting in your plugin

{
    "platform": "PLC",
    "ip": "192.168.10.1",
    "rack": 0,
    "slot": 1,
    "enablePolling": true,
    "enablePush": true,
    "enableControl": true,
    "port": 8080,
    "accessories": [
        {
            "accessory": "PLC_LightBulb",
            "name": "LuceSala",
            "manufacturer": "normal light bulb",
            "db": 1,
            "get_On": 0,
            "set_On": 0.1,
            "set_Off": 0.2
        }
    ]
}

Feilner, 2020-11-01:
The terminal could be found here but maybe it's only present on the docker variant I'm using:
[image]

S7-1200 and S7-1500 CPU access requires access using "Slot 1"(your config is correct) 
So it's likly the PLC that prevents you from accessing.
You must disable optimized block access (in TIA portal) for the blocks you are using. In addition, you must "Enable GET/PUT Access" in the 1200/1500 controller in TIA Portal. 

Check also the link from the Readme:
http://snap7.sourceforge.net/snap7_client.html#1200_1500

Nrigoni, 2020-11-01:
I have tryed to reinstall all and seams that homebridge in connected to PLC, I have paired it in the home app of iPhone too and for a few minute the acessory "LuceSala" turn on and off (only as GUI because on PLC the Q0.0 associate to state of bool in the DB remain always ON) after few minute in the HomeApp on iPhone the accessory not show the actual state but N/D and in web UI I can't turn it on or off what I'm doing wrong?

I attaching here all the print screen of homebridge settins and TIA settings

Thanks a lot for the support

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Nrigoni, 2020-11-01:
Sorry, here the config.json
[config.txt](https://github.com/Feilner/homebridge-plc/files/5470363/config.txt)

Feilner, 2020-11-01:
Hello,

Your configuration seems to be absolutely correct. 
From the log you hand provided I can see that the connection is also successful. 

For sure you have tried to control the configured Lamp und you could not see any change within the DB. 
Is there any program running on the PLC that resets the bits before you can see it?

Maybe you could enable the debug mode to see more information.

Feilner, 2020-11-01:
[image]

You should see something like:
[11/1/2020, 9:31:01 AM] [PLC] [Arbeitszimmer] get On: 0 (getBit DB6094DBX0.0)
[11/1/2020, 9:31:06 AM] [PLC] [Arbeitszimmer] set On: true (setOnOffBit DB6094DBX1.1)
[11/1/2020, 9:31:29 AM] [PLC] [Arbeitszimmer] set On: false (setOnOffBit DB6094DBX1.0)

Nrigoni, 2020-11-01:
Now it is working perfectly from web interface I have made the error to set and reset as you have said. 

Now I don't undestand why I can't enstablish a connection with Home App ... The accessory not response for a few minute has worked now I can't get back the connection

Feilner, 2020-11-01:
**Great to here that homebridge-plc plugin works for you now.**
I will close the issue now. If you have further questions please open a new issue.

Regarding the connection between the Home App and homebridge I would recommend to try to reset Homebridge and connect the Home-App again.
[image]

Nrigoni, 2020-11-01:
I have changed get_ON in DB from 0.0 to 0.3 

when I press once the Lamp button on web interface I get this

[1/11/2020, 09:47:40] [PLC] [LuceSala] get On: 0 (getBit DB1DBX0.3)
[1/11/2020, 09:47:42] [PLC] [LuceSala] set On: true (setOnOffBit DB1DBX0.1)
[1/11/2020, 09:47:42] [PLC] [LuceSala] get On: 0 (getBit DB1DBX0.3)
[1/11/2020, 09:47:43] [PLC] [LuceSala] get On: 0 (getBit DB1DBX0.3)

it's different from yours why? 

thanks

Feilner, 2020-11-01:
Because of 4 things: 
1. I have triggernd from the home app, This one does not query the current state with "get_On" after switching on.
2. I have a program running that switches the device on and reset afterward the bit that requested the switch on.
3. I switched also off
4. I have manually filtered the log currently I have 72 PLC devices configured I also have the push configuration running that pushes a state change via http request from the PLC to the plugin.

Here's a log from my test Installation (just one device):
[image]

Feilner, 2020-11-01:
1. I used the Home App and not the Homebridge UI to trigger. The Home App does not query the state after switching.
2. I have a program running that actually switching the device on after requesting the switch on and reset the request bit.
3. I have manually copied the lines from the log. As I have currently 72 PLC devices in my productive configuration and also use the push of changes as http request from the PLC to the plugin.

Here is a log from a test environment with on/off/on from Homebridge UI with only one device within the configuration:
[image]

Feilner, 2020-11-01:
1. I used the Home App and not the Homebridge UI to trigger. The Home App does not query the state after switching.
2. I have a program running that actually switching the device on after requesting the switch on and reset the request bit.
3. I have manually copied the lines from the log. As I have currently 72 PLC devices in my productive configuration and also use the push of changes as http request from the PLC to the plugin.

Feilner, 2020-11-01:
I used the Home App and not the Homebridge UI to trigger. The Home App does not query the state after switching.
I have a program running that actually switching the device on after requesting the switch on and reset the request bit.
I have manually copied the lines from the log. As I have currently 72 PLC devices in my productive configuration and also use the push of changes as http request from the PLC to the plugin.

Nrigoni, 2020-11-02:
Perfect, thank you Feilner. Now I have buy the raspberry because the homebridge give me some problem running on windows. If I need your hel I will open a new issue.

Feilner, 2020-11-02:
Your are welcome.
By the way if you have a NAS may can run HomeBridge as docker container on it.

closed 2020-11-02 by Feilner

Nrigoni, 2020-11-05:
Hello Feiner, I have setting up the Homebridge with Raspberry and in the Home app I can see the bridge online. But if I connect the ether cable in log of Homebridge UI I can see that it is connected to PLC but in the Home app it goes offline. Can you help me? thanks

Feilner, 2020-11-05:
Hello,

I have fully understood your question. 
So you have set up the Raspi and connected to the Homa APP with Homebridge. 

You see in the log that the PLC is connected with the Homebridge-plc plugin. 

What you mean with the Ethernetkabel I did not understand. 

Can you control the Acessories from Homebridge?

/Feilner

Nrigoni, 2020-11-05:
Sorry... yes,

I have set up Ras an connected to Home App and I see that the accessory Bridge is online (in the home app)
In the log of Homebridge I see that the PLC in connected with the plugin but now when the PLC is connected to the plugin the bridge accessory in the Home app appear offline.

Nrigoni, 2020-11-05:
and in the mean time I receive this error every 1 minutes 

[05/11/2020, 20:28:04] [Homebridge UI] [homebridge] Failed to check registry.npmjs.org for updates: "connect EHOSTUNREACH 104.16.23.35:443" - see https://git.io/JJSz6 for help.
[05/11/2020, 20:28:07] [Homebridge UI] [homebridge-config-ui-x] Failed to check registry.npmjs.org for updates: "connect EHOSTUNREACH 104.16.17.35:443" - see https://git.io/JJSz6 for help.
[05/11/2020, 20:28:07] [Homebridge UI] [homebridge-plc] Failed to check registry.npmjs.org for updates: "connect EHOSTUNREACH 104.16.25.35:443" - see https://git.io/JJSz6 for help.

Nrigoni, 2020-11-05:
now I have done this code in terminal 

sudo systemctl restart homebridge 

at restart for very few minutes all works via home app I will command the accessory and see the log in Homebridge UI but suddenly in Home App all goes offline and start to write the error that I have sent in the commend before.

I think that two things are related

sorry for all this comment

Feilner, 2020-11-05:
```
[05/11/2020, 20:28:07] [Homebridge UI] [homebridge-plc] Failed to check registry.npmjs.org for updates: "connect EHOSTUNREACH 104.16.25.35:443" - see https://git.io/JJSz6 for help.

```

That seems to be the homepridge—UI that locks if there are updates for the installed plugins and indicates that your RASPI can't reach the internet. 

When the home app shows the accessories as offline can you still control the PLC from Homebridge?

For me it seems to be a issue either with your network settings or with homebridge. 

I'm using the homebridge docker container.

Nrigoni, 2020-11-05:
now seems to work, I have change a ethernet settings on raspberry 

thanks for support

now I start to develop the software for accessories hope to not disturb you more.

Nrigoni, 2020-11-06:
Hello Feilner, I have a question when I turn on a light with LightBulbe for example from Home app but after this light is turn off manually from a button connected to an input of the PLC there is a possibility tu turn it off on the home app too? So I mean change the state in the GUI of home app? or it's not possible because the home app don't read the actual state from PLC?

Feilner, 2020-11-06:
Hello,

see the readme: https://github.com/Feilner/homebridge-plc#update-of-values

In principle there a some possible options.
1. you open close the home app or change the room that will force the home app to query the actual state again
2. you enable the polling mode to query the actual state from the PLC in fixed intervals. 
3. you enable the push mode where you push a updated value from the PLC to the homebridge plugin by an HTTP request.

I'm using option 3 in my home automation.

Best regards,

Oliver

Nrigoni, 2020-11-06:
wow perfect,

so do you use the TSEND_C block in TIA? for senting the HTTP request

thanks

Feilner, 2020-11-06:
Yes, but as I have a S7-300 compatible PLC it is 
 * TCON
 * TSEND
 * TREV
 * TDISCON

I have head already the HTTP request implemented as I'm using this also to send push notifications via Pushover, get Weather data, Switch on/off the light of my Miele hood in the kitchen.

Nrigoni, 2020-11-06:
ok perfect,

and I have to push the get on value in case of light bulbe right?

Feilner, 2020-11-06:
Yes when changing the status you would need to push the changed value.

I have implemented a FC with an ANY pointer to put changed values in a FIFO to send.

Here is the example:

FUNCTION "pushValue_value" : Bool 
{ S7_Optimized_Access := 'FALSE' }
VERSION : 0.1
   VAR_INPUT
      i_pAny : Any;   // Variable to be pushed
   END_VAR

   VAR_TEMP
      isOk : Bool;
      valueBoolean : Bool;
      index : Int;
      anyPtr : "any_anyPtr_t";
      valueReal : Real;
      valueDint : DInt;
   END_VAR

BEGIN
#isOk := "any_isDbPointer"(i_pAny := #i_pAny);
IF #isOk THEN
  #isOk := "fifo_addIndex"(i_pElement := "IDB_pushValue".m_fifoBuffer[0], i_pStorage := "IDB_pushValue".m_fifoBuffer, io_instance := "IDB_pushValue".m_fifo);
  IF #isOk THEN
    #index := "fifo_getIndex"(io_instance := "IDB_pushValue".m_fifo);
    #anyPtr := "any_decodeAnyPtr"(i_pAny := #i_pAny);
   
    "IDB_pushValue".m_fifoBuffer[#index].db := "DINT_TO_STRING"(IN := WORD_TO_DINT(#anyPtr.dbNumber));
   
    IF #anyPtr.typ = "ANY_TYPE_BOOL" THEN
      #valueBoolean := false;
      #valueDint := 0;
      #valueReal := 0.0;
      #valueBoolean := WORD_TO_BLOCK_DB(#anyPtr.dbNumber).DX(#anyPtr.offset, #anyPtr.offsetBit);
      "IDB_pushValue".m_fifoBuffer[#index].value := "BOOL_TO_STRING"(IN := #valueBoolean);
      "IDB_pushValue".m_fifoBuffer[#index].offset := CONCAT(IN1 := "DINT_TO_STRING"(IN := #anyPtr.offset), IN2 := '.', IN3 := "DINT_TO_STRING"(IN := #anyPtr.offsetBit));
    ELSE
      #valueDint := 0;
      "IDB_pushValue".m_fifoBuffer[#index].offset := "DINT_TO_STRING"(IN := #anyPtr.offset);
      "IDB_pushValue".m_fifoBuffer[#index].value := "DINT_TO_STRING"(IN := #valueDint);
      IF
        (*u8*)#anyPtr.typ = "ANY_TYPE_BYTE" OR #anyPtr.typ = "ANY_TYPE_CHAR" THEN
        #valueDint := WORD_TO_BLOCK_DB(#anyPtr.dbNumber).DB(#anyPtr.offset);
        "IDB_pushValue".m_fifoBuffer[#index].value := "DINT_TO_STRING"(IN := #valueDint);
      ELSIF
        (*u16*)#anyPtr.typ = "ANY_TYPE_WORD" OR
        (*s16*)#anyPtr.typ = "ANY_TYPE_INT"       THEN
        #valueDint := WORD_TO_BLOCK_DB(#anyPtr.dbNumber).DW(#anyPtr.offset);
        "IDB_pushValue".m_fifoBuffer[#index].value := "DINT_TO_STRING"(IN := #valueDint);
      ELSIF
        (*u32*)#anyPtr.typ = "ANY_TYPE_DWORD" OR #anyPtr.typ = "ANY_TYPE_TIME" OR
        (*s32*)#anyPtr.typ = "ANY_TYPE_DINT"  THEN
        // Note: DWORD is not allowed to be bigger then 0x7FFFFFF as it would bekome negative
        #valueDint := WORD_TO_BLOCK_DB(#anyPtr.dbNumber).DD(#anyPtr.offset);
        "IDB_pushValue".m_fifoBuffer[#index].value := "DINT_TO_STRING"(IN := #valueDint);
      ELSIF #anyPtr.typ = "ANY_TYPE_REAL" THEN
        #valueReal := DWORD_TO_REAL(WORD_TO_BLOCK_DB(#anyPtr.dbNumber).DD(#anyPtr.offset));
        "IDB_pushValue".m_fifoBuffer[#index].value := "REAL_TO_STRF"(IN := #valueReal, N := 1);
      ELSE
        // "ANY_TYPE_TOD", "ANY_TYPE_DT", (32bit)
        // "ANY_TYPE_DATE", "ANY_TYPE_S5TIME" (16bit)
        //"ANY_TYPE_STRING", "ANY_TYPE_BLOCK_FB" ,"ANY_TYPE_BLOCK_FC", "ANY_TYPE_BLOCK_DB", "ANY_TYPE_BLOCK_SDB", "ANY_TYPE_COUNTER", "ANY_TYPE_TIMER":
        #isOk := false;
      END_IF;
    END_IF;
  END_IF;
END_IF;
#pushValue_value := #isOk;
 
END_FUNCTION

Nrigoni, 2020-11-09:
Thanks Oliver for the example provided. 

I have une more question por pushing value via HTTP request I have to put PLC IP in the same class of HOMEBRIDGE?

at this moment I have PLC with this IP address : 192.168.10.1

and homebridge ethernet port: 192.168.10.80

but homebridge is connected to the local network via wifi and it has this IP: 192.168.1.59

so the question is HTTP request has been done to the 192.168.10.80 or 192.168.1.59?

thanks

Feilner, 2020-11-09:
Hello,

I would guess:
When push changed values from the PLC you should use the IP that in the same network as the PLC (192.168.10.80)
When testing from the browser you should use the IP of the local network (192.168.1.59)
When testing from command line where homebride is installed you could also use the loop back address (127.0.0.1 <http://127.0.0.1:8888/?control&db=6094&offset=1.1&value=1>)

Please check also the port that there is no collision with homebridge UI.

Here is an example from the command line where homebridge is installed:

        {
            "platform": "PLC",
            "ip": "10.10.10.32",
            "rack": 0,
            "slot": 2,
            "enablePush": true,
            "enableControl": true,
            "port": 8888,
            "accessories": [
{
                    "accessory": "PLC_LightBulb",
                    "name": "Arbeitszimmer",
                    "manufacturer": "DG",
                    "db": 6094,
                    "set_On": 1.1,
                    "set_Off": 1.0,
                    "get_On": 0
                }
}]

curl "http://127.0.0.1:8888/?control&db=6094&offset=1.1&value=1" 
curl "http://127.0.0.1:8888/?control&db=6094&offset=1.1&value=0"

The log with debug info looks:
[11/9/2020, 2:35:38 PM] [PLC] [HTTP Control] Received control request for accessory:6094 offset:1.1 value:1 
[11/9/2020, 2:35:38 PM] [PLC] [Arbeitszimmer] Control On:1
[11/9/2020, 2:35:38 PM] [PLC] [Arbeitszimmer] set On: true (setOnOffBit DB6094DBX1.1)
[11/9/2020, 2:35:40 PM] [PLC] [HTTP Control] Received control request for accessory:6094 offset:1.1 value:0
[11/9/2020, 2:35:40 PM] [PLC] [Arbeitszimmer] Control On:0

Best regards

/Oliver