Feilner/homebridge-plc #14: homebridge-plc plugin can't be loaded after update of docker image

Feilner, 2020-11-02:
The log states the folling
```
[11/2/2020, 11:26:45 PM] Loading 4 platforms...
[11/2/2020, 11:26:45 PM] [Config] Initializing config platform...
[11/2/2020, 11:26:45 PM] [Config] Running in Service Mode
[11/2/2020, 11:26:45 PM] Error loading platform requested in your config.json at position 2
[11/2/2020, 11:26:45 PM] Error: The requested platform 'PLC' was not registered by any plugin.
    at PluginManager.getPluginForPlatform (/usr/local/lib/node_modules/homebridge/src/pluginManager.ts:224:15)
    at /usr/local/lib/node_modules/homebridge/src/server.ts:385:37
    at Array.forEach (<anonymous>)
    at Server.loadPlatforms (/usr/local/lib/node_modules/homebridge/src/server.ts:372:27)
    at Server.start (/usr/local/lib/node_modules/homebridge/src/server.ts:151:29)
    at cli (/usr/local/lib/node_modules/homebridge/src/cli.ts:80:10)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge/bin/homebridge:17:22)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
[11/2/2020, 11:26:45 PM] Got SIGTERM, shutting down Homebridge...
[11/2/2020, 11:26:50 PM] [HB Supervisor] Homebridge Process Ended. Code: 143, Signal: null
[11/2/2020, 11:26:55 PM] [HB Supervisor] Restarting Homebridge...
[11/2/2020, 11:26:55 PM] [HB Supervisor] Starting Homebridge with extra flags: -I -P /homebridge/node_modules
[11/2/2020, 11:26:57 PM] [HB Supervisor] Started Homebridge v1.2.3 with PID: 1106
[11/2/2020, 11:26:58 PM] Loaded config.json with 0 accessories and 4 platforms.
[11/2/2020, 11:26:58 PM] ---
[11/2/2020, 11:26:58 PM] Loaded plugin: homebridge-camera-ffmpeg@3.0.3
[11/2/2020, 11:26:58 PM] Registering platform 'homebridge-camera-ffmpeg.Camera-ffmpeg'
[11/2/2020, 11:26:58 PM] ---
[11/2/2020, 11:26:58 PM] Loaded plugin: homebridge-dummy@0.4.1
[11/2/2020, 11:26:58 PM] Registering accessory 'homebridge-dummy.DummySwitch'
[11/2/2020, 11:26:58 PM] ---
[11/2/2020, 11:26:58 PM] Loaded plugin: homebridge-harmony@1.4.0
[11/2/2020, 11:26:58 PM] Registering platform 'homebridge-harmony.HarmonyHubWebSocket'
[11/2/2020, 11:26:58 PM] ---
[11/2/2020, 11:26:58 PM] ====================
[11/2/2020, 11:26:58 PM] ERROR LOADING PLUGIN homebridge-plc:
[11/2/2020, 11:26:58 PM] Error: Could not locate the bindings file. Tried:
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/build/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/build/Debug/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/build/Release/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/out/Debug/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/Debug/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/out/Release/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/Release/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/build/default/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/compiled/14.15.0/linux/x64/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/addon-build/release/install-root/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/addon-build/debug/install-root/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/addon-build/default/install-root/node_snap7.node
 → /homebridge/node_modules/homebridge-plc/node_modules/node-snap7/lib/binding/node-v83-linux-x64/node_snap7.node
    at bindings (/homebridge/node_modules/homebridge-plc/node_modules/bindings/bindings.js:126:9)
    at Object.<anonymous> (/homebridge/node_modules/homebridge-plc/node_modules/node-snap7/lib/node-snap7.js:8:45)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/homebridge/node_modules/homebridge-plc/index.js:6:13)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
[11/2/2020, 11:26:58 PM] ====================
```

Feilner, 2020-11-02:
The problem is the snap7 lib. After the update of the docker container the snap7 library is not installed any more.
Steps to fix:
1. Deinstall homebridge-plc plugin
2. Install homebridge-plc plugin again
3. Restart

That´s it!

closed 2020-11-02 by Feilner

sebfia, 2020-12-28:
Unfortunately the problem wasn't solved for me with deinstalling and installing.

Feilner, 2020-12-28:
I'm running also in within a docker container (oznu/homebridge:latest)
After updating I have head similar issues but at my side a deinstall and install solved the issue.
The root cause seems to be that after the update of the docker container the dependency towards node-snap7 or its dependencies is not correctly resolved. 

I would  recommend to try to install node-snap7 by command line (I'm not sure if you should install with or without -g)
npm install -g node-snap7

Feilner, 2020-12-28:
I have just updated the docker from version from 2020-12-06 01:29:31 to version from 2020-12-18 12:41:14
without having any issues. (Just used recreate within portainer.io and tagend download latest image)

sebfia, 2020-12-31:
Thank you for your help! After installing node-snap7 globally it works like charm. Thanks for the great project!!!