Feilner/homebridge-plc #20: Seperate DB config files for get and set

leonschwanitz, 2020-11-26:
Is it possible to configure separate dbs for setting and querying? e.g. set_on: DB105.DBX10.0 and for status DB108.DBX10.0?

If you could put that in I would be super happy. Or should I commit it and put it in? :)

Feilner, 2020-11-27:
Hello Leon,

No this is currently not possible. 

I thought about it when I started with the plugin if I should support different locations for everything including markers, input, and output. But decided not to do it as it would made one one hand side the configuration more complex and on the other side I thought it would be a rather seldom use case. As you have typically one function block e.g. for handling lights which has all required like local Interface for control from plc as well as remote interface for homebridge and the status inside. 

For example I do use instance DBs only (no multi instances) for accessories that can be controlled by homebridge.  By this I the offsets are the same for all instances and I only need to change the DB number in the configuration.

For the normal and polling mode it would be possible to extend it with medium effort. For the Push and Control it would be much harder as here the DB is used for performance reasons quite to find the correct accessory.

What would be your suggestion how to do the configuration and handling the Push and Control?

Best regards,

Oliver

leonschwanitz, 2020-11-30:
Thanks for the fast reply. We have restructured the DBs in our PLC and working now with a single DB. Thank you anyways :)

closed 2020-11-30 by leonschwanitz