Feilner/homebridge-plc #27: PLC temeperaturesensor don´t work

juspa1978, 2021-01-31:
When tryed to use PLC TemperatureSensor i don´t get accessory to homekit room screen 
"accessory": "PLC_TemperatureSensor",
					"name": "Temperature",
					"db": 9,
					"get_CurrentTemperature": 16
					"enablePolling": true,
					"pollInterval": 60'
If i modify "accessory": "PLC_TemperatureSensor", -> "accessory": "Temperature", then i get message on homeapp that accessory is not currently supported

Feilner, 2021-01-31:
Can you post your complete config?
Have you added it inside a platform config?
E.g.
{
"platform": "PLC",
"ip": "10.10.10.32",
"rack": 0,
"slot": 2,
"enablePolling": true,
"defaultPollInterval": 15,
"distributePolling": true,
"enablePush": true,
"enableControl": true,
"port": 8888,
"accessories": [
{
Here
}
]

juspa1978, 2021-01-31:
"platform": "PLC",
"ip": "192.168.0.146",
"rack": 0,
"slot": 1,
"enablePolling": true,
"defaultPollInterval": 15,
"distributePolling": true,
"enablePush": true,
"enableControl": true,
"port": 8888,
"accessories": [

Exsamble ”plc switch” works fine. Problem only with temperature sensor

juspa1978, 2021-01-31:
[image]
So homeapp gives message ”homeapp not currently supporting this device”

Feilner, 2021-01-31:
> F6C224BF-8052-4E39-865F-87280E599109 <https://user-images.githubusercontent.com/76985784/106381304-70118880-63c0-11eb-9495-47f6801aab0f.jpeg>

The picture from the home App?
The logo does not fit. I have never saw this. 
It should look like the attached. 
Are you using homebridge-UI? Can you have a look how the accessory is displayed there?
Did you head a look at the log wir enabled debug mode?

juspa1978, 2021-01-31:
At hoobs accessories page i can see temperature correctly.

Feilner, 2021-01-31:
Ok now we get more light in the mystery.
You are not using homebridge you are using hoobs.

I have short downloaded and tested hoobs. For this I used the hoobs docker container.
I can confirm that the plc plugin works also works with hoobs and all accessories are correctly shown at the hoobs accessory page.

I also connected hoobs to the home app and all my accessories including the PLC_TemperatureSensor are working fine.

// to test it I executed the following on a ubuntu machine (I added no storage so it had no history at restert)
docker run --net=host hoobs/hoobs

Best regards,

Oliver

juspa1978, 2021-01-31:
Thanks for support problem was still on my hoobs over all config.... i had "accessories[]" 2 times on hoobs config... weird thing is that plc_switch in my config worked just fine. 

So actual issue was my self! 🤥

closed 2021-02-02 by Feilner