Feilner/homebridge-plc #32: PLC communication issue

hbothner, 2021-04-17:
During testing my PLC_TemperatureSensor, i recognized a strange problem. As soon as i enable the "Homebridge-PLC v1.0.28" plugin with the proper configuration to show my living room temperature, i encounter a problem with my programming device. At this point the communication  is no longer possible with the plc. In detail i try to watch a plc db to compare the temperature values between homebridge and the siemens plc programming laptop. 

I have the problem vice versa too! If i start the PLC watch before i activate the homebridge plugin. The plugin will not get a connection.

Is that a configuration issue? How can i avoid this behaviour? Which communication protocol is used to enable a connection to the plc?

My setup :
Simatic 300 (316-2DP)
CP 343-1 lean

Feilner, 2021-04-18:
I'm not sure about the capabilities of the CP 343-1 lean but according to the technical data it supports 4 concurrent connections. Therefore this should not be an issue.

As next step I have created a project with 316-2DP and CP 343-1 lean. 
And yes it is a configuration issue of the 316-2DP. 
By default the CPU is configured to only accept one PG (Programmiergerät [Programming Console]) Connection. As Snap7 which is used as base for the communication uses  PG connection as default. This conflicts.
You have to increase the number of PG Connections.
[image]

hbothner, 2021-04-18:
I have updated the "PG-Kommunikation" parameter and it works fine!

closed 2021-04-18 by hbothner

Feilner, 2021-04-21:
I have added a new option which would also solve your connection problem without changing the configuration of the PLC. By switching from PG to OP Communication. At least when you have no HMI-Panel connected that already uses the one default OP connection resource.

`communicationOP`: (optional) when set to `true` OP-Communication is used instead of PG-Communication

hbothner, 2021-04-21:
That sounds good, i will try at the weekend.