Feilner/homebridge-plc #47: Failed to create a child bridge

Shikaban, 2024-04-08:
After updating and fixing the config file with the new names as reported on the changelog everything is working as intended but if i try to use the GUI of the plugin or try to create an isolated child bridge this are the error that appear:
- The first one is when i open the setting page
- Second one is when i try to save any configuration
Am i doing something wrong?

[image]

Feilner, 2024-04-08:
Hello,
I just gave it a try and moved the homebridge-plc into a child bridge using the GUI and it worked as expected. 

Could you send me your complete homebridge config.json file to have a look on it?

Shikaban, 2024-04-08:
```

    "bridge": {
        "name": "HomebridgePi",
        "username": "0E:14:52:D7:D8:48",
        "port": 51928,
        "pin": "516-07-929",
        "advertiser": "avahi"
    },
    "accessories": [],
    "platforms": [
        {
            "name": "Config",
            "port": 8581,
            "platform": "config",
            "scheduledBackupDisable": true
        },
        {
            "platform": "PLC",
            "ip": "192.168.15.51",
            "rack": 0,
            "slot": 1,
            "enablePolling": true,
            "defaultPollInterval": 10,
            "distributePolling": true,
            "enablePush": true,
            "enableControl": true,
            "port": 8888,
            "accessories": [
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Soffitto armadio",
                    "enablePolling": true,
                    "db": 11,
                    "get_On": 8,
                    "set_On_Set": 0,
                    "set_On_Reset": 0.1
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Matrimoniale",
                    "enablePolling": true,
                    "db": 11,
                    "get_On": 8.1,
                    "set_On_Set": 0.2,
                    "set_On_Reset": 0.3
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Soffitto giallo",
                    "enablePolling": true,
                    "db": 11,
                    "get_On": 8.2,
                    "set_On_Set": 0.4,
                    "set_On_Reset": 0.5
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Specchio giallo",
                    "enablePolling": true,
                    "db": 11,
                    "get_On": 8.3,
                    "set_On_Set": 0.6,
                    "set_On_Reset": 0.7
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Anticamera",
                    "enablePolling": true,
                    "db": 11,
[... 318 lines cut ...]
                },
                {
                    "accessory": "PLC_Valve",
                    "name": "Irri zona 2",
                    "db": 12,
                    "enablePolling": true,
                    "ValveType": 1,
                    "get_Active": 10.2,
                    "set_Active_Set": 4.3,
                    "set_Active_Reset": 4.4
                },
                {
                    "accessory": "PLC_SecuritySystem",
                    "name": "Allarme",
                    "db": 12,
                    "enablePolling": true,
                    "get_SecuritySystemCurrentState": 23,
                    "get_SecuritySystemTargetState": 23
                }
            ]
        }
    ]
}

````

Feilner, 2024-04-09:
Hello,
the pasted config did not contain a sub bridge.

I have stiped down the config to one accessory and tried it to be working fine.
Beside I'm using no auth I have not seen any mayor difference. 

Could you try with the below config?

```
{
    "bridge": {
        "name": "HomebridgePi",
        "username": "0E:14:52:D7:D8:48",
        "port": 51928,
        "pin": "516-07-929",
        "advertiser": "avahi"
    },
    "accessories": [],
    "platforms": [
        {
            "name": "Config",
            "port": 8581,
            "auth": "none",
            "platform": "config",
            "scheduledBackupDisable": true
        },
        {
            "platform": "PLC",
            "ip": "192.168.15.51",
            "rack": 0,
            "slot": 1,
            "enablePolling": true,
            "defaultPollInterval": 10,
            "distributePolling": true,
            "enablePush": true,
            "enableControl": true,
            "port": 8888,
            "accessories": [
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Soffitto armadio",
                    "enablePolling": true,
                    "db": 11,
                    "get_On": 8,
                    "set_On_Set": 0,
                    "set_On_Reset": 0.1
                }
            ],
            "_bridge": {
                "username": "0E:64:D4:9D:0B:8C",
                "port": 54462
            }
        }
    ]
}
```

Here is the corresponding log:

```
[4/9/2024, 11:01:15 AM] [Homebridge UI] Homebridge restart request received
[4/9/2024, 11:01:15 AM] [Homebridge UI] UI / Bridge settings have not changed; only restarting Homebridge process
[4/9/2024, 11:01:15 AM] [Homebridge UI] Sending SIGTERM to Homebridge
[4/9/2024, 11:01:15 AM] Got SIGTERM, shutting down Homebridge...
[4/9/2024, 11:01:15 AM] Got SIGTERM, shutting down child bridge process...
[4/9/2024, 11:01:20 AM] [HB Supervisor] Homebridge Process Ended. Code: 143, Signal: null
[4/9/2024, 11:01:25 AM] [HB Supervisor] Restarting Homebridge...
[4/9/2024, 11:01:25 AM] [HB Supervisor] Starting Homebridge with extra flags: -I -P /var/lib/homebridge/node_modules -D --strict-plugin-resolution
[4/9/2024, 11:01:25 AM] [HB Supervisor] Started Homebridge v1.7.0 with PID: 2421
[4/9/2024, 11:01:25 AM] Loaded config.json with 0 accessories and 2 platforms.
[4/9/2024, 11:01:25 AM] Loaded 0 cached accessories from cachedAccessories.
[4/9/2024, 11:01:25 AM] ---
[4/9/2024, 11:01:25 AM] Loaded plugin: homebridge-plc@2.0.3
[4/9/2024, 11:01:25 AM] Registering platform 'homebridge-plc.PLC'
[4/9/2024, 11:01:25 AM] ---
[4/9/2024, 11:01:25 AM] Loading 2 platforms...
[4/9/2024, 11:01:25 AM] [PLC] Initializing PLC platform...
[4/9/2024, 11:01:25 AM] [PLC] Initializing child bridge 0E:64:D4:9D:0B:8C
[4/9/2024, 11:01:25 AM] Publishing bridge accessory (name: HomebridgePi, publishInfo: {
  username: '0E:14:52:D7:D8:48',
  port: 51928,
  pincode: '***-**-***',
  category: 2,
  bind: undefined,
  mdns: undefined,
  addIdentifyingMaterial: true,
  advertiser: 'avahi'
}).
Setup Payload:
X-HM://0024BNP15HGXX
Enter this code with your HomeKit app on your iOS device to pair with Homebridge:
                       
    ┌────────────┐     
    │ 516-07-929 │     
    └────────────┘     
                       
[4/9/2024, 11:01:25 AM] Homebridge v1.7.0 (HAP v0.11.1) (HomebridgePi) is running on port 51928.
[4/9/2024, 11:01:26 AM] [homebridge-plc] Launched child bridge with PID 2432
[4/9/2024, 11:01:26 AM] Registering platform 'homebridge-plc.PLC'
[4/9/2024, 11:01:26 AM] [homebridge-plc] Loaded homebridge-plc v2.0.3 child bridge successfully
[4/9/2024, 11:01:26 AM] Loaded 0 cached accessories from cachedAccessories.0E64D49D0B8C.
[4/9/2024, 11:01:26 AM] [homebridge-plc] Connecting to 10.10.10.32 (0:2) PG-Communication
[4/9/2024, 11:01:26 AM] [homebridge-plc] Connected to 10.10.10.32 (0:2) PG-Communication
[4/9/2024, 11:01:26 AM] [homebridge-plc] Add PLC accessories...
[4/9/2024, 11:01:26 AM] [homebridge-plc] [1/1] Soffitto armadio (PLC_LightBulb)
[4/9/2024, 11:01:26 AM] [homebridge-plc] Polling enabled interval 10s. First polling is done in 2s
[4/9/2024, 11:01:26 AM] Initializing platform accessory 'Soffitto armadio'...
[4/9/2024, 11:01:26 AM] [homebridge-plc] Enable polling...
[4/9/2024, 11:01:26 AM] [homebridge-plc] Init done!
[4/9/2024, 11:01:26 AM] Publishing bridge accessory (name: homebridge-plc, publishInfo: {
  username: '0E:64:D4:9D:0B:8C',
  port: 54462,
  pincode: '***-**-***',
  category: 2,
  bind: undefined,
  mdns: undefined,
  addIdentifyingMaterial: true,
  advertiser: 'avahi'
}).
[4/9/2024, 11:01:26 AM] [homebridge-plc] Enable push and control server...
[4/9/2024, 11:01:26 AM] [homebridge-plc] Listening on port 8888
[4/9/2024, 11:01:26 AM] Homebridge v1.7.0 (HAP v0.11.1) (homebridge-plc) is running on port 54462.
[4/9/2024, 11:01:28 AM] [homebridge-plc] [Soffitto armadio] Execute polling...
[4/9/2024, 11:01:28 AM] [homebridge-plc] [Soffitto armadio] get On: 0 (getBit DB6049DBX0.0)
[4/9/2024, 11:01:38 AM] [homebridge-plc] [Soffitto armadio] Execute polling...
[4/9/2024, 11:01:38 AM] [homebridge-plc] [Soffitto armadio] get On: 0 (getBit DB6049DBX0.0)
[4/9/2024, 11:01:48 AM] [homebridge-plc] [Soffitto armadio] Execute polling...
[4/9/2024, 11:01:48 AM] [homebridge-plc] [Soffitto armadio] get On: 0 (getBit DB6049DBX0.0)
```

Shikaban, 2024-04-09:
Hi thanks for the reply , I've manage to understand what was causing the issue.

I was running home assistant with node 18 and before upgrading the plc plugin I've migrated to node 20 of course I have to run the npm recompile to let the plugin work.
Problem was caused by not having restarted the entire raspberry, I was doing some tracing of an humidity sensor and in that moment I could not turn off the Pi, and I've only restarted the home assistant service thinking it was enough.

Your configuration is working and even my configuration, after the complete reboot I can now create the bridge without any issue.

Fortunately I've got some spare Pi's that I can use to test and I've emulated nearly exactly what I've done yesterday.

Thanks a lot for your help and also for your work!!!

Do you know if there is any possibility in the future to extend the communication with symbolic access avoiding the old S7 comm?