Feilner/homebridge-plc #51: The HomeKit app does not update the state of accessories

Vodnik1989, 2024-08-20:
The HomeKit app does not update the state of accessories. Only when I close and reopen the app. In the browser Homebridge updates the status well.

Feilner, 2024-08-20:
When you close/open the home App it queries all accessories and therefore update the state of the accessories. 

In order to update the state when the Home App is open or to use automatizations  you have to setup the polling mode of the plugin. 
Or better setup the push functionality. 
in that case the plugin provides an http server and expects that the plc inform about a changed  state by an http request.

Vodnik1989, 2024-08-20:
But I tried without success. It works in the web browser, so the PLC is responding, but there is probably an error in the application or the phone.

Feilner, 2024-08-20:
When the homebridge webpage updates but not the App it‘s likely the you News an control center. 

https://support.apple.com/en-en/102557

closed 2025-01-12 by Feilner