Feilner/homebridge-plc #52: accessory out of compliance error

blaster22, 2024-12-27:
Hello,
The HomeKit app don't upgrade accessories and stuck also when remove and add all again.

with this JSON all works fine. When i try to add other PLC temperature sensor homekit application stuck and don't work.
`{
    "bridge": {
        "name": "Homebridge 6990",
        "username": "0E:A9:8F:4E:A6:1B",
        "port": 51402,
        "pin": "129-22-950",
        "advertiser": "bonjour-hap"
    },
    "accessories": [],
    "platforms": [
        {
            "name": "Config",
            "port": 8581,
            "platform": "config",
            "lang": "it"
        },
        {
            "ip": "192.168.10.100",
            "rack": 0,
            "slot": 1,
            "communicationOP": false,
            "enablePolling": true,
            "distributePolling": true,
            "accessories": [
                {
                    "accessory": "PLC_Switch",
                    "name": "Spegni tutto",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 0.2,
                    "set_On": 0.2
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Corridoio",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 8.3,
                    "set_On_Set": 5.7,
                    "set_On_Reset": 6
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Binario living",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 6.4,
                    "set_On_Set": 2.1,
                    "set_On_Reset": 2.2
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Applique divano",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 6.5,
                    "set_On_Set": 2.3,
                    "set_On_Reset": 2.4
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Faretti ingresso",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 6.6,
                    "set_On_Set": 2.5,
                    "set_On_Reset": 2.6
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "LED ingresso",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 6.7,
                    "set_On_Set": 3,
                    "set_On_Reset": 3.1
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Lampada tavolo",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 7.1,
                    "set_On_Set": 3.3,
                    "set_On_Reset": 3.4
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "LED tavolo",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 7.6,
                    "set_On_Set": 4.5,
                    "set_On_Reset": 4.6
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Faretti living",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 7.7,
                    "set_On_Set": 4.7,
                    "set_On_Reset": 5
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Balcone cucina",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 8.1,
                    "set_On_Set": 5.3,
                    "set_On_Reset": 5.4
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "LED sotto TV",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 8.2,
                    "set_On_Set": 5.5,
                    "set_On_Reset": 5.6
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "Camera matrimoniale",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 7.4,
                    "set_On_Set": 3.7,
                    "set_On_Reset": 4
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "TV camera matrimoniale",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 7.5,
                    "set_On_Set": 4.1,
                    "set_On_Reset": 4.2
                },
                {
                    "accessory": "PLC_LightBulb",
                    "name": "LED sotto cucina",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_On": 70.2,
                    "set_On_Set": 70,
                    "set_On_Reset": 70.1
                },
                {
                    "accessory": "PLC_TemperatureSensor",
                    "name": "Esterna",
                    "db": 15,
                    "manufacturer": "Company",
                    "enablePolling": true,
                    "get_CurrentTemperature": 10
                }
            ],
            "platform": "PLC"
        }
    ]
}`

simply add other "PLC_TemperatureSensor" for stuck all.
There is a limitation on query on PLC? or limitation on homekit?

On "Accessory" on Homebridge i see temperature but on Homekit not. Whats i can do?
 

thanks

Feilner, 2024-12-28:
Hallo,

Can you describe a bit more what you mean with the home app stucks?

Does the state update when you close and reopen the home app?

Do you have an AppleTV as control center?

Are you sure that the temperature sensor is not showing up? As temperature sensors are only shown in home app when open the room. 
[image]
[image]

Have you already enabled the "Debug mode" of homebridge in the error case and head a look at the log file?

If not please do so and send the log.

blaster22, 2024-12-28:
Hello,

if i add more than one temperature sensor HomekitApp stuck like in photo, but in homebridge i see all fine. (see photo)

[image]
[image]
[image]
[image]

From log i don't see problems.

[log.txt](https://github.com/user-attachments/files/18267846/log.txt)

blaster22, 2024-12-28:
And i saw another problem. I used PLC garagedooropener feature, also in this case, i see in HomeBridge status without problem and on  Homekit is always in "updating"

for now i use only app on Iphone. I bought AppleTV yesterday, i'm waiting to receive it.

JSON is:
                    "accessory": "PLC_GarageDoorOpener",
                    "name": "Garage",
                    "db": 15,
                    "manufacturer": "Elmod",
                    "enablePolling": true,
                    "get_CurrentDoorState": 71,
                    "get_TargetDoorState": 8.4,
                    "set_TargetDoorState": 72
[log garage.txt](https://github.com/user-attachments/files/18267860/log.garage.txt)

Feilner, 2024-12-29:
I have first looked at the issue with the PLC_GarageDoorOpener:
- I can reproduce the issue
- It is related to your configuration
- get_TargetDoorState has to be of type `Byte` therefore no `Bit` address like `8.4` is allowed. 
- please change e.g to "get_TargetDoorState": 73,

I will now look on the other issue.

Feilner, 2024-12-29:
For you issue with the temperature sensor I have could also not see an issue from the log file.
Having more temperature sensors is no is normal I have seven in my environment.

To be honest currently I have no glue what the issue could be therefore my questions and suggestions are just narrow down the issue.

The only thing I coud can see is that the two accessories are polled at the same time. Which is absolutely fine. For better traceability of the issue you could set the defaultPollInterval to a number > as the number of accessories you have defined.

```
[12/28/2024, 10:04:17 PM] [PLC] [LED ingresso] Execute polling...
[12/28/2024, 10:04:17 PM] [PLC] [Camera matrimoniale] Execute polling...
[12/28/2024, 10:04:17 PM] [PLC] [LED ingresso] get On: 0 (getBit DB15DBX6.7)
[12/28/2024, 10:04:17 PM] [PLC] [Camera matrimoniale] get CurrentTemperature: 19.899999618530273 (getReal DB15DBD26)
```
To exclude that long response times from your PLC are the issue I have done a new 2.0.7 release to log also the execution time of the PLC request. Could you please retest with this please?

When you remove the Temperature sensor form the json does it then work instantly again?

What is your setup?
- In which environment is Homebrige executed
- What PLC are you using

I have seen various issues with the Home app connecting to Homebridge when Homebridge MAC address was changed. In my case it was caused by getting a new MAC address when restarting the container.
Are you sure that you have always the same MAC address?

blaster22, 2024-12-29:
> I have first looked at the issue with the PLC_GarageDoorOpener:
>
> * I can reproduce the issue
> * It is related to your configuration
> * get_TargetDoorState has to be of type `Byte` therefore no `Bit` [...]

Ok, perfect. Now it work fine! Thank you!

blaster22, 2024-12-29:
> For you issue with the temperature sensor I have could also not see an issue from the log file. Having more temperature sensors is no is normal I have seven in my environment.
>
> To be honest [...]

Hello,
I don't know why but now it works well. I just added one by one with plugin configuration and not directly in JSON configuration. By the way, i also add you log with new version of plugin.

Hardware is:
- PLC S7-1200 CPU1214C
- Raspberry PI 3 B+ (distro Diet-Pi)

Thank you too much for your help!

[log v2.txt](https://github.com/user-attachments/files/18270317/log.v2.txt)

closed 2025-01-07 by Feilner