Feilner/homebridge-plc #55: Add power meter function

blaster22, 2025-01-02:
Hello,
is possible add a power meter like accessory?

It is REAL value (like temperature) but with different unit of measurement, and icon. 

Thank you too much

Feilner, 2025-01-03:
Which Service Type is it?

I do not find a Power Meter:
https://developers.homebridge.io/#/

Feilner, 2025-01-12:
Currently the Apple Home App has no support for energy metering.
In fact there are 3rd party apps like EVE that bring the support. There is a plug in that supports the showing the energy meter of a Shelly 3EM within the EVA app.
https://github.com/produdegr/homebridge-3em-energy-meter

Currently I have no plans to integrate such a functionality but would of cause accept a PR.

closed 2025-01-12 by Feilner